cleanflight/cleanflight-configurator #466: Error when disconnecting an old version of firmware

MauroMombelli, 2017-04-19:
Hi,

i jus received a little emachine gx95, it uses a sp3evo but with an old version of firmware (or so say the configurator); then i disconnected to upload a newer firmware, but the main windows remain blank aside from the top and lateral menu.

log:

[oldSP3evo.txt](https://github.com/cleanflight/cleanflight-configurator/files/938119/oldSP3evo.txt)

hydra, 2017-04-29:
try latest version, should be fixed now.

BadRaven, 2017-05-04:
I had this too TODAY on the same quad AND on another quad with the same FC. Had to install an old version of configurator to adjust PIDS which was all that was required.

MauroMombelli, 2017-05-04:
Your issue seems different as this one the configurator is usable unless
you do a specific sequence of action. Can you get the error log from the js
console?  Maybe is better if youbopen a different issue and in case it will
be flag as duplicate.

closed 2020-05-24 by stale[bot]