cleanflight/cleanflight-configurator #467: Fixed Wing + Naze (et. al) Issues w/ Solutions

jprucks, 2017-04-20:
**I am not sure where this issue originates, I'm not an expert, but I cannot track down the cause.**

I have a Skywalker 1680 with Hobbyking's Illuminati32 FCB, Ublox GPS, and factory 9x receiver (PWM). 
>>Inputs/outputs
On the FCB, I have two motors powered from the M1 output on the FCB (no diff. thrust), no output on M2. M3, M4, M5, M6 go to AIL-L, AIL-R, RUD, and ELEV servos. 
On the FCB, my receiver inputs are routed to CH1, 2, 5, 6 (+7&8 for switches/pots). CH3&4 are used for the GPS. The GPS module's magnetometer is routed to the FCB's (serial?) port. 

>>I was running CF 1.1x.x and an older version of Configurator. Everything worked fine with PASSTHROUGH turned on, even the GPS with my PWM rx inputs (I didn't have to map anything, I just turned on the UART port II, IIRC). Without it on, I got servo drift on throttle. This is because the standard CF mixer uses stabilized RC inputs, I believe. **This should be changed for the fixed-wing modes, and maybe added as an option.**

**ISSUE 1**
>>I was forced to upgrade due to some problems. I'm now using CF2.0.0 with a new Configurator. 

M1 shows the correct throttle position and the motors spool up correctly. But NONE of my servos work. S4, S5&6 (AIL), S7 correctly show stick outputs. **For some reason, none of the output pins show any output on an oscilloscope, and the servos do not move or resist movement.** Servos have power and twitch on startup, like normal. I have double-checked every option I think is related, but the steep learning curve is holding me back. 
**ISSUE 1 SOLUTION**
Resource remapping is required, but the SERVO numbers do not match the numbers in the Motors tab of Configurator. 
They match the servo numbers in the servo mixer, plus 1. _Example: if the Tx shows the pitch servo as Servo 4 in the Motors tab, but is used as Servo 0 in SMIX, then "resource MOTOR 3 B06" is cleared with "resource MOTOR 3 NONE" and then mapped "resource SERVO 1 B06"._ I believe this is referred to as the "rule" or "index" in SMIX. 

>>This process is overcomplicated and confusing. I was actually using Betaflight when I solved it, but I do not think it makes a difference. Effort should be put into simplifying these options in a linear way (Step 1, 2, 3, etc). with common reference numbers. 
Maybe Cleanflight could implement a step-by-step SERVO/MOTOR & INPUT/OUTPUT SETUP tab in the GUI, now that the CLI "resource" commands are functional and usable?

**ISSUE 2**
I think this might be related, but I do not know. My GPS no longer works, although all of the transmitter's inputs are correctly received and mapped by the FCB. I'm using AE12TR34 channel map, and I'm not sure how to use the "resource" option to tell the firmware that 1&2 are serial Tx/Rx from the GPS.

McGiverGim, 2017-04-20:
The GPS support has been removed for NAZE for space reasons, you must compile your own firmware to enable it again.

jprucks, 2017-04-20:
> The GPS support has been removed for NAZE for space reasons, you must compile your own firmware to enable it again.
Do you know when (which version) it was removed? It's necessary for what I'm using the plane for. I have very basic programming knowledge and I doubt I could figure it out on my own.

McGiverGim, 2017-04-20:
In version 2.0. You must return to 1.4.2.

jprucks, 2017-04-20:
Cleanflight Configurator won't connect to the board with older firmware. That is what started this entire issue.

jprucks, 2017-04-20:
Nevermind, thanks for the help though. I switched to the newest NAZE iNav firmware and got everything working again, with GPS.

hydra, 2017-04-29:
great, thanks for the update.

closed 2017-04-29 by hydra