cleanflight/cleanflight-configurator #469: No ESC response

RYH1996, 2017-04-21:
Hi there

I have updated on the latest cleanflight version 2.x.
I wasn`t happy with this version so I flashed the older version on my Naze32 and the older version on my laptop. Since then I have no response from my ESCs not a single beep.
Could anybody  give me an advice?

Thanks for response
sincerely Yves

grahouu, 2017-04-22:
I have a same problem, I bought a JJRC P130 but when I update the firmware (1.12.0 to 2.X), my transmitter no longer wants to communicate with my rc, I have a flysky fs-i6 (V1.1)... Sorry, i am a novice in this field

hydra, 2017-04-29:
wrong place to ask for support, please use slack or forums.  check your motor output protocol, you probably have it on the new default oneshot instead of pwm.

closed 2017-04-29 by hydra