cleanflight/cleanflight-configurator #483: mag_declination units are wrong

McGiverGim, 2017-05-24:
Seeing the cleanflight documentation, for a magnetic declination of -1º10' the command is:
`set mag_declination=-110`
But if I use the configurator to insert this, the CLI shows a value of:
```
# get mag_declination
mag_declination = -1100
Allowed range: -18000 - 18000
```
I can try to make a PR to solve this, but I want to be sure that the documentation is right and the configurator is wrong before. Can someone confirm that the documentation is right?

mikeller, 2017-05-24:
Well spotted. Looks like the culprit is here:

https://github.com/cleanflight/cleanflight-configurator/blob/22b82cc0fd39bb901d6a57cfb97861c981136396/js/msp/MSPHelper.js#L312

for what I can tell, the change that switched from sending the truncated value to sending the full value never made it into Betaflight:

https://github.com/cleanflight/cleanflight/blob/master/src/main/fc/fc_msp.c#L1075

Not sure how to best solve this. In reality, sending the truncated value does not make much sense, since it's not even a decimal (and hence not scaled), but a concatenation of degrees and minutes:

https://github.com/cleanflight/cleanflight/blob/master/src/main/sensors/compass.c#L151-L152

My hunch would be to break protocol compatibility (like it was done with 1.18) and switch from sending the truncated value to sending the full value. @hydra thoughts?

McGiverGim, 2017-05-24:
I've seen a mess of values in the FC.

The `compassConfig()->mag_declination` seems to store the mag_declination * 10 (for example, for the 1º10' it stores 1100). 
When sending this value using the MSP, in the `fc_msp` it is multiplied (when receiving) or divided by 10 (when sending). But when using the CLI the real value of `compassConfig()->mag_declination` is sended.

As far as I can see, the `compassConfig()->mag_declination` now stores internally the value * 10. So the `compass.c` that you said maybe never worked. Maybe this is the cause because the navigation with GPS in betaflight is useless?

McGiverGim, 2017-05-24:
Ok, understood, so the documentation is right and the configurator must send '-110' for the -1º10' and is sending '-1100' by error. This is what I thinked first but after seeing the code I wasn't sure ;)

closed 2020-05-24 by stale[bot]