cleanflight/cleanflight-configurator #488: Serial Port usage

arnav-gudibande, 2017-06-09:
Currently, I have a simple python script that opens a serial port, runs a couple of functions, and then closes the port.

```
import serial

ser = serial.Serial('/dev/ttyUSB0')
print(ser.name) # example function
ser.close()
```

However, after I run this script, cleanflight-configurator is not able to successfully establish a connection on on `/dev/ttyUSB0`. The log shows `Serial port successfully opened with ID: 2` and then timeouts after 10 seconds with `Serial port successfully closed`.

As of now, the only way to actually establish a connection after running that script is to fully reboot the computer. Why is cleanflight unable to reopen a serial port?

stronnag, 2017-06-09:
Perhaps because you leave the port in an indeterminate state. If you run
````
stty sane < /dev/ttyUSB0
````
does that improve things?. fyi I have no problems running arbitrary home made programs (cli save/restore, bb log downlaoder) against CF compatible FCs and subsequently using the configurator.

arnav-gudibande, 2017-06-09:
Interesting, `stty sane` fixes the issue

arnav-gudibande, 2017-06-09:
Thanks that worked!

closed 2017-06-09 by arnav-gudibande

mikeller, 2017-06-11:
Little addendum: `stty sane -F /dev/<usb device>` is a preferable form of the same command.