cleanflight/cleanflight-configurator #493: Arm switch not activating when within range

JaredBledsoe, 2017-06-21:
(Similar issue as another issue that was closed a while back, but none of their "fixes" fixed mine)
I'm using a ImpulseRc Helix and am not able to get it to arm. The receiver is calibrated properly with 1000min and 2000max, and all the settings are within range to properly arm. Below is a dump of my settings, any help would be greatly appreciated!
# Betaflight / IMPULSERCF3 3.1.7 Apr 3 2017 / 22:04:16 (e1c4b5c)
name -
resource BEEPER 1 C15
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B01
resource MOTOR 4 B00
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A15
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
mixer QUADX

mmix reset

servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
[... 381 lines cut ...]
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
rateprofile 0
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

mikeller, 2017-06-21:
Hi @JaredBledsoe.

Please don't scatter gun post your questions across multiple projects. Select the one that is most appropriate, and post there.

Also, see my comment on your post in https://github.com/cleanflight/cleanflight-configurator/issues/493.

JaredBledsoe, 2017-06-21:
My bad, will do!

closed 2017-06-21 by JaredBledsoe