cleanflight/cleanflight-configurator #498: Cleanflight configurator V2.1.0-RC1 has problem

E120103016, 2017-07-15:
Hello there:
   The cleanflight configurator V2.1.0-RC1 has a problem. Cause After updating the firmware v2.1.0-RC1, use the cleanflight configurator V2.1.0-RC1. Can not make the settings. Jump directly to the CLI. Is there a problem.

trekki1978, 2017-07-30:
I'm having the same issue:
[image]

E120103016, 2017-07-31:
The current configuration device V2.1.0 is the same problem. I do not know how to solve. As trekki1978 said the same problem do not know how to inform the changes.TKS!

joemonti, 2017-08-02:
I had the same issue and believe this fix resolves it -- #503

trekki1978, 2017-08-02:
The file features.js in the folder "...\AppData\Local\Junemike\User Data\Default\Extensions\enacoimjcgeinfnnnpajinjgmkahmfgb\2.1.0_0\js" seems to be very different from the file changed in the issue #503.

trekki1978, 2017-08-04:
I could solve this issue and start the configurator after I deleted the "," at the end of line 75:
"{bit: 18, group: 'other', name: 'OSD'}**,**"

I could use the windows editor to change the file FEATURES.JS in folder:
C:\Users\...\AppData\Local\Junemike\User Data\Default\Extensions\enacoimjcgeinfnnnpajinjgmkahmfgb\2.1.0_0\js

Maybe this helps the one or the other user who wants to use the latest CF configurator.

E120103016, 2017-08-05:
Has been resolved. Thank you very much.

closed 2020-05-24 by stale[bot]