cleanflight/cleanflight-configurator #502: Cleanflight configurator V2.1.0 has problem

E120103016, 2017-07-31:
The current configuration device V2.1.0 is the same problem. I do not know how to solve. As trekki1978 said the same problem do not know how to inform the changes.TKS!

trekki1978, 2017-08-04:
I could solve this issue and start the configurator after I deleted the "," at the end of line 75:
"{bit: 18, group: 'other', name: 'OSD'}**,**"

I could use the windows editor to change the file FEATURES.JS in folder:
C:\Users\...\AppData\Local\Junemike\User Data\Default\Extensions\enacoimjcgeinfnnnpajinjgmkahmfgb\2.1.0_0\js

Maybe this helps the one or the other user who wants to use the latest CF configurator.

E120103016, 2017-08-05:
Thanks! You provide information

E120103016, 2017-08-05:
已經解決可以使用非常感謝.

E120103016, 2017-08-05:
Has been resolved. Thank you very much.

7ocb, 2017-08-08:
@trekki1978 thanks, that helped me too.

closed 2020-05-24 by stale[bot]