cleanflight/cleanflight-configurator #504: Not selecting servo correctly - Corrupts servo data on "save".

cj7hawk, 2017-08-09:
A problem I've noticed with Cleanflight 2.1.0 configurator and 2.1.0 firmware together - Servo assignment is broken. 

This problem appears to originate with the configurator. When selecting servos, the configuration is always set to Channel -90 - which doesn't exist. This is clearly a fault. 

example to reproduce fault; if I select Servo0 as Aux4 so as to mix gimbal and input, and "save" what is saved to the CLI configuration is this;

servo 0 1000 2000 1500 100 -90

And when I re-enter the SERVOS tab, the box is no longer ticked.

What should have happened however is that the following should have been saved;

servo 0 1000 2000 1500 90 7 

So the last digit should be 7 instead of -90.

On changing this manually under CLI, and saving, then returning to the SERVOS menu, the correct tickbox for AUX4 is now selected, however and change and "save" through the configurator will once again change the channel to -90.

This is a pretty major bug for anyone using a gimbal or Servo Tilt.  I hope the above information is helpful in locating the bug and resolving it.

Thanks
David.

Mmarsian, 2017-10-05:
I can confirm this problem, same for me.

bkacjios, 2017-10-06:
Yep yep! Same issue here. Thankfully the command line in the OP got my servo working.

closed 2020-05-24 by stale[bot]