cleanflight/cleanflight-configurator #506: Tricopter support for NAZE-32 broken

icebreaker-ch, 2017-09-01:
I updated my NAZE-32 firmware from 1.2.4 to 2.1.0 for my tricopter to use the latest cleanflight-configurator software.
I was not able to get the Servo on S1 to work. It seemed, like one of the ESC is served on channel 1, instead of the Yaw-Servo.
In the "Motors" display, the yaw movements are shown as channel 6 but connecting the Servo to channel 6 also was unsuccessful.
I installed the old cleanflight configurator version 1.2.4 and (re-)loaded the 1.2.4 firmware to the NAZE-32 and everything was ok again (Servo on channel 1, Motors on channels 3,4,5).
I also tried firmware 2.0.1 on the new cleanflight-configurator which also did not work.

My configuration is attached.
[CLFL_cli_20170901_162952.txt](https://github.com/cleanflight/cleanflight-configurator/files/1271640/CLFL_cli_20170901_162952.txt)

mikeller, 2017-09-03:
Your configuration seems to be from an old version of the firmware. This looks like the firmware update did not work at all, and instead just mangled your configuration.

icebreaker-ch, 2017-09-03:
In fact the file represents the old configuration which I tried to load into the new software.
But even after a full erase and new full setup of the configuration, the behavior was the same; Servo on channel 1 seems to be controlled as an ESC, not as the yaw servo (No reaction on rudder stick but on moving the copter in roll and pitch axes). For a tricopter, the connections on the ESC port should be as follows:
CH1: Yaw Servo
CH2: -
CH3: Motor 1
CH4: Motor 2
CH5: Motor 3

I Think this is exactly the same problem as described here: https://github.com/cleanflight/cleanflight/issues/2775
Seems the mapping has been changed compared to Version 1 or the servo is not assigned at all.
I will try to reconfigure the resource mapping as described in issue 2775.

icebreaker-ch, 2017-09-03:
The following mapping seems to work for my NAZE32:

```
resource servo 1 A08
resource servo 2 A11
resource motor 1 B06
resource motor 2 B07
resource motor 3 B08
```
I would appreciate, if this mapping is set on using a Tricopter configuration (as it was in V1)

icebreaker-ch, 2017-09-03:
The following remapping works for my NAZE32 (CLI commands:

```
resource servo 1 A01
resource servo 2 A11
resource motor 1 B06
resource motor 2 B07
resource motor 3 B08
```

I would appreciate, if this mapping is set on choosing a Tricopter configuration.

mikeller, 2017-09-03:
@icebreaker-ch: Good to see that you managed to find the solution for your problem! The problem with resource mappings for different mixers is that they are entirely dependend on pin allocation for the different outputs, and hence different for every board. (i.e. on NAZE, motor pins 1 & 2 share a common timer, and 3 - 5 share another one, so the split between motors and servos has to be 1, 2 / 3, 4, 5.) In cleanflight 1, which supported 2 dozen or so different boards, the definition of the resource mappings for different  mixers for all of these boards was a horrible mess. With over 100 boards supported in cleanflight 2, maintaining these mappings in firmware is simply not feasible. At some stage we will be looking into adding and maintaining mappings in the configurator, but even this will require a lot of work, and unfortunately a lot of the board vendors are not exactly forthcoming with support for the target definitions for their boards...

icebreaker-ch, 2017-09-03:
Thanks a lot for your work.

closed 2017-09-03 by icebreaker-ch