cleanflight/cleanflight-configurator #514: Cleanflight Configurator 2.1.0 Firmware Flash Stops if Full Chip Erase is Enabled

BillAllman, 2018-01-08:
SP Racing F4 EVO brand new. All drivers loaded and working correctly. Start Cleanflight and select firmware flash. Select SP Racing F4 EVO for FC model and current build 2.2.0 and download. Turn on No Reboot Sequence and Full Chip Erase. Hold boot loader button on FC and plug in USB. Top right shows DFU mode. Click flash firmware. Erase bar starts and when it starts flashing it just stops and nothing happens. The log shows serial device closed before it flashes. If I disable Full Chip Erase and click Flash Firmware it goes through the whole flash and reboots. I have tried other boot loader drivers but it still does the exact thing. I tried this in Betaflight and it works just fine.

mikeller, 2018-01-08:
That's a long standing configurator bug: betaflight/betaflight-configurator#493

BillAllman, 2018-01-15:
Thank you for letting me know. What problems does it cause if it can't do a full chip erase? I used Betaflight and it worked great but how will that effect Cleanflight if I flash it back?

mikeller, 2018-01-15:
That's no problem - most likely the change in firmware will be detected during the reboot after flashing, and a reset of the configuration will be triggered. To make sure the configuration has been reset, just do a 'Reset to defaults' in the Setup tab.

closed 2020-05-24 by stale[bot]