cleanflight/cleanflight-configurator #518: OSD Elements switch doesn't work, others are fiddly

tonipalmgren, 2018-03-25:
Using the standalone configurator.

When trying to toggle the "Elements" switch on the OSD tab, nothing happens. Logo switch works well.

Still on the OSD tab, if I change either "Source" or "Precision", and then change the other of the two, both of the fields will be cleared. If I then check this on the Chrome-based configurator (without saving), the first timer will be empty.

Failsafe stage 2 cannot be set in the Chrome-based configurator once set in the standalone version. (Settings for the menu won't appear at all.)

On the Receiver tab, if I change the "'Stick Low' threshold" by clicking on either down or up in a row, it will sooner or later start "clicking" on the other value (the other button will start to flash, but will not change any value.) Same happens on other tabs, too.

Using "cleanflight-configurator-installer_2.2.0-RC1_win32" (standalone), and version 2.1.0 (Chrome-based).

SP Racing F3 NEO/OSD, with latest firmware (2.2.0 (rel. 3/3/2018)). Did a full erase while updating.

hydra, 2018-03-26:
thanks for the feedback, will investigate.

hydra, 2018-03-26:
Likely 1,2 are caused by firmware and configurator mismatch (i.e. new firmware with old configurator). unable to replicate when using firmware 2.3.0-RC1 and configurator 2.2.0-RC1

tonipalmgren, 2018-03-27:
1 and 2 work in the Chrome extension, which is old (2.1.0), but not in the standalone version, which is newer. I'm using the latest standalone version available from https://github.com/cleanflight/cleanflight-configurator/releases, and the latest firmware it gave me. Shouldn't they be compatible? If the latest configurator can offer a firmware that is not compatible with the configurator, things get messy.

It's odd that these settings work in the old Chrome version but not the newer standalone version. If this was caused by a firmware/configurator version mismatch, shouldn't the issue be the other way around (i.e. settings not working in the old Chrome extension)?

hydra, 2018-03-28:
`latest firmware it gave me` isn't very specific as it depends on what board you have and if you chose to see unstable releases.  exactly which firmware were you trying?  latest 2.3.0-RC1? - https://github.com/cleanflight/cleanflight/releases should be ok.

There's probably a few more version checks that need adding to address some of the issues you were having however.   They wouldn't necessarily be compatible, that's not how software works - you have to specifically make it work 😄

tonipalmgren, 2018-03-28:
Standalone shows the version as:
Running - OS: Windows, Chrome: 63.0.3239.132, Configurator: 2.2.0.

Installed from:
cleanflight-configurator-installer_2.2.0-RC1_win32.exe

Firmware on NEO, OSD end EVO:
CLFL_v2.2.0 - SPRACINGF3EVO - 03-03-2018 20:51 (stable)

I can't find a more recent standalone version. Chrome extension is the one you get from the Play store. I guess I need to download the firmware manually.

tonipalmgren, 2018-03-30:
I can confirm that the OSD Elements switch works with the standalone configurator and firmware v2.3.0-RC1 which is the newest firmware available from https://github.com/cleanflight/cleanflight/releases at the moment.

However, when clicking on the "Power & Battery" tab, all I get is "Waiting for data...". Hitting "Disconnect" and then "Connect", results in a "Failed to open serial port" error. I need to cycle the power to be able to connect again. Same happens in both the standalone and the Chrome extension versions of the configurator. Connecting the main battery doesn't help.

Tested on an SP Racing F3 NEO.

closed 2020-05-24 by stale[bot]