cleanflight/cleanflight-configurator #526: Running tests: CONFIG is not defined

drnic, 2019-02-07:
First time developer, following the README to run tests I ran `npm test` but get the error output below. Is there some additional configuration or documentation for how I run any unit/integration tests?

I did not have any flight controllers connected when running `npm test`.

```
$ npm test

> cleanflight-configurator@2.5.0 test /Users/drnic/Projects/flightcontrollers/cleanflight-configurator
> karma start test/karma.conf.js

07 02 2019 10:39:13.786:WARN [watcher]: All files matched by "/Users/drnic/Projects/flightcontrollers/cleanflight-configurator/node_modules/chai/chai.js" were excluded or matched by prior matchers.
07 02 2019 10:39:13.832:INFO [karma]: Karma v3.0.0 server started at http://0.0.0.0:9876/
07 02 2019 10:39:13.833:INFO [launcher]: Launching browser ChromeHeadlessNoSandbox with unlimited concurrency
07 02 2019 10:39:13.838:INFO [launcher]: Starting browser ChromeHeadless
07 02 2019 10:39:14.171:INFO [HeadlessChrome 0.0.0 (Mac OS X 10.14.2)]: Connected on socket ydLKPKFpdTGWhHH3AAAA with id 13161
HeadlessChrome 0.0.0 (Mac OS X 10.14.2) TABS.cli input exit upon cleanup clears cliBuffer first FAILED
	ReferenceError: CONFIG is not defined
	    at GUI_control.content_ready (src/js/gui.js:291:9)
	    at src/js/tabs/cli.js:179:13
	    at n.fn.init.sinon.stub.callsFake (test/tabs/cli.js:112:17)
	    at Object.invoke (node_modules/sinon/pkg/sinon.js:409:32)
	    at n.fn.init.functionStub (node_modules/sinon/pkg/sinon.js:3173:47)
	    at Function.invoke (node_modules/sinon/pkg/sinon.js:2770:51)
	    at n.fn.init.proxy [as load] (node_modules/sinon/pkg/sinon.js:2641:60)
	    at Object.TABS.cli.initialize (src/js/tabs/cli.js:54:19)
	    at Context.done (test/tabs/cli.js:251:22)
HeadlessChrome 0.0.0 (Mac OS X 10.14.2): Executed 13 of 13 (1 FAILED) (0.006 secs / 0.02 secs)
TOTAL: 1 FAILED, 12 SUCCESS
npm ERR! Test failed.  See above for more details.
```

mikeller, 2019-02-07:
@drnic: Yes, looks like tests are failing for the cleanflight configurator. Maybe try Betaflight configurator (https://github.com/betaflight/betaflight-configurator). Tests are working there, and most changes originate there and are then pulled into cleanflight configurator.

drnic, 2019-02-07:
Thanks @mikeller

closed 2019-02-07 by drnic