cleanflight/cleanflight-configurator #528: MSP Communication Problem

berkkduman, 2019-03-11:
I want to communicate with my fc via MSP without CleanFlight. I can easily spin motors in receiver tab with enabling controls. Because i want to build autonomous drone, i want to send information to fc via my raspberry pi over serialport. I can easily arm and set fc raw values with nodejs but motors dont spin. I tried to analyse cleanflight configurator to find how cleanflight handle with that issue. I couldnt find anything. I guess, i am missing something to send to fc with MSP. Can anyone help me to find this problem?

mikeller, 2019-03-11:
https://github.com/cleanflight/cleanflight-configurator/blob/master/src/js/msp/MSPHelper.js

eman291, 2019-12-01:
Raspberry Pi may be used in conjunction with a L293D Chip to control the rotation of the DC Motor. (PWM) Pulse width modulation controls the speed. 

- https://www.rs-online.com/designspark/raspberry-pi-controlling-a-motor-via-gpio

- https://github.com/betaflight/betaflight-tx-lua-scripts

closed 2020-05-24 by stale[bot]