cleanflight/cleanflight-configurator #543: No Linux builds available

gcormier, 2020-08-16:
Release notes mention three packages for Linux - a deb, rpm and zip file. None are availalable on releases section since 2.4.0.

https://github.com/merbanan/rtl_433/issues/1464#issuecomment-671526419

gcormier, 2020-09-15:
Still an issue.

neuhaus, 2020-11-08:
Please reopen

hydra, 2021-02-08:
As noted in https://github.com/cleanflight/cleanflight-configurator/issues/545#issuecomment-775128842 the issue is as follows:

I recall trying, and then running into some of the usual linux pain, the result of the build was this:
```
[13:47:21] Error: Command failed: dpkg-deb --build ./release//cleanflight-configurator_2.6.0_amd64
dpkg-deb: warning: parsing file './release//cleanflight-configurator_2.6.0_amd64/DEBIAN/control' near line 7 package 'cleanflight-configurator':
 missing description
dpkg-deb: warning: parsing file './release//cleanflight-configurator_2.6.0_amd64/DEBIAN/control' near line 7 package 'cleanflight-configurator':
 missing maintainer
dpkg-deb: error: control directory has bad permissions 777 (must be >=0755 and <=0775)
```

I'm gonna have another shot at it now and see what I can do, EDIT: still doesn't build.

I'm using WSL + Ubuntu to build.  It basically seems like the debian package generator can't run under WSL and/or there are bugs and/or magic runes which I don't know.

If anyone can help it'd be greatly appreciated.

hydra, 2021-02-08:
I managed to build it by git cloning the the source to the WSL `~/src` which builds, results in these messages:

```
[13:52:57] dpkg-deb: building package 'cleanflight-configurator' in './release//cleanflight-configurator_2.6.0_amd64.deb'.

[13:52:57] dpkg-deb: warning: parsing file './release//cleanflight-configurator_2.6.0_amd64/DEBIAN/control' near line 7 package 'cleanflight-configurator':
 missing description
dpkg-deb: warning: parsing file './release//cleanflight-configurator_2.6.0_amd64/DEBIAN/control' near line 7 package 'cleanflight-configurator':
 missing maintainer
dpkg-deb: warning: ignoring 2 warnings about the control file(s)
[13:52:57] Finished 'release_linux64_deb' after 32 s
[13:52:57] Finished 'release' after 1 min
```

The runes for the build process were:

```
cd ~/src/
git clone /mnt/d/Documents/dev/chrome/cleanflight-configurator
cd cleanflight-configurator/
git tag
git checkout CLFL_v2.6.0
git reset --hard
nvm use 10.23
yarn install
yarn gulp release --linux64
ls -al release/
```

I've added the packages to the release, please try them and report back.  I'll unpin this so the bot closes this issue automatically if there's no feedback.

closed 2021-03-18 by stale[bot]