cleanflight/cleanflight-configurator #545: No Linux build available of v2.6.0 release in June

neuhaus, 2020-11-08:
This is a duplicate of #543 which was closed by the bot.

There are still no Linux packages available for version 2.6.0.

The last version of the Linux binaries is from 2018.

neuhaus, 2020-12-08:
Bad bot

neuhaus, 2021-01-08:
Bad bot

neuhaus, 2021-02-07:
Bad bot

neuhaus, 2021-02-07:
@hydra Hi, can you build the binary or does someone else do this?

hydra, 2021-02-08:
@neuhaus Hey, sorry for late reply to this, I recall trying, and then running into some of the usual linux pain, the result of the build was this:
```
[13:47:21] Error: Command failed: dpkg-deb --build ./release//cleanflight-configurator_2.6.0_amd64
dpkg-deb: warning: parsing file './release//cleanflight-configurator_2.6.0_amd64/DEBIAN/control' near line 7 package 'cleanflight-configurator':
 missing description
dpkg-deb: warning: parsing file './release//cleanflight-configurator_2.6.0_amd64/DEBIAN/control' near line 7 package 'cleanflight-configurator':
 missing maintainer
dpkg-deb: error: control directory has bad permissions 777 (must be >=0755 and <=0775)
```

I'm gonna have another shot at it now and see what I can do.

hydra, 2021-02-08:
Gonna close this, and re-open the original #543

closed 2021-02-08 by hydra

mikeller, 2021-02-08:
@hydra: If you are running on a Windows host it is probably the easiest if you use VirtualBox or similar and install a small-ish linux VM to use for building the linux packages.